waveshareteam/Waveshare-ESP32-components #5: Issues regarding P4-Nano | Support Referral

Ncouch64, 2025-01-14:
Hello, I was referred to GitHub issues by waveshare support for a particular problem I am encountering with my P4-Nano board.

I have two primary concerns, and believe I am pretty close to solving one of my issues but I will include it here.

1. I have been attempting to get the speaker to work with the Nano Kit C, I have tried the audio example included in the waveshare wiki, the audio example included as part of ESP-IDF and have even been in contact with the ESP devs of the audio driver to try and get things working. So far I have been able to get the project to build, flash and run but when its running I am getting no output from the speaker. I have tried different configs, verifying pin assignments, and also verifying that PA_CTRL was set to GPIO_53 and set to HIGH. Despite all this I have not been able to get any audio from the speaker. Is there potentially something that I am missing here? Any recommendations on what I could attempt from here to get things working?

2. The final concern has to do with the CSI-Camera functionality. Originally I was not able to get anything from the camera but have sense gotten some of the examples working. P4 using the different camera driver from previous boards due to the mipi connection has created some confusion on the best way to actually utilize the component for doing something as simple as capturing a single image and performing basic processing to it. Are there any examples that showcase this more clearly? I saw the CSI-DSI example from the waveshare wiki but I don't have an LCD screen and it seems a little complex for just starting out.

Thanks for the assistance here!

WaveshareCloud, 2025-01-16:
Hi @Ncouch64 ,
Thank you for your feedback. This library is a component adapter library. When you use the i2s example provided by Waveshare wiki, the test can be completed after the development board is powered off and powered on

CSI camera features of ESP32-P4-NANO You can use SD card to record video or capture images and output from USB-A to computer as an extended camera feature, you can refer to this example
https://github.com/espressif/esp-iot-solution/tree/master/examples/camera

Ncouch64, 2025-01-16:
> Hi [@Ncouch64](https://github.com/Ncouch64) , Thank you for your feedback. This library is a component adapter library. When you use the i2s example provided by Waveshare wiki, the test can be [...]

Hello and thank you for the response! Is there another spot I should take this issue post? I am only posting here based on the referral from waveshare support website.

As for the i2s example, I have just tried using the waveshare i2s example again. I flashed the board, did a power cycle, and still dont receive any audio from the speaker. This is the output I am getting below:

[image]

Like I mentioned previously, I have tried many solutions including recommendations coming straight from ESP devs and still have not gotten things to work correctly. I am really not sure what the issue is at this point.

As for the camera component, this link https://github.com/espressif/esp-iot-solution/tree/master/examples/camera is a good example of working with esp32-cam component but the P4 uses the new esp_video component and is the recommended method for utilizing the camera moving forward, as described by ESP devs. The new component is still in active development and is based on V4L2, I find it to be overly confusing and am working on figuring things out. For comparison, esp32-cam basic example is maybe 100 lines of code that, while initially confusing, is pretty easy to parse and work with. The basic example for esp-video is over 300 lines of code and all over the place. Still learning...

WaveshareCloud, 2025-01-17:
Sorry to hear that, but it's not actually a hardware issue, nor is it a problem with the sample program we provided. Our development board has a screen connection in the factory and then test music playback, have you modified the code? Here is a tested firmware that you can flash and test. This firmware works fine on my ESP32-P4-NANO

[ESP32-P4-NANO-I2S_Demo_250117.zip](https://github.com/user-attachments/files/18447594/ESP32-P4-NANO-I2S_Demo_250117.zip)

WaveshareCloud, 2025-01-20:
This issue has nothing to do with this component library, and no feedback has been received for a long time. Therefore, this issue is closed. If there are new problems, you can raise them again or reopen this issue

closed 2025-01-20 by WaveshareCloud