waveshareteam/Waveshare-ESP32-components #8: 4" Round display Black screen?

Platin21, 2025-01-26:
Hey,

Just want to inform that seemingly the LCD-Driver doesn't work.
I get no picture i also changed the value that was suggested for another display sort of maybe to try if i can see anything else sadly no luck either. 

I did see for a short glimps some color bar at the bottom but i think this was more a glitch then actually any real rendering. 

The code i used to setup the panel was:
```c
ESP_LOGI(TAG, "MIPI DSI PHY Powered on");
    esp_ldo_channel_handle_t ldo_mipi_phy = NULL;
    esp_ldo_channel_config_t ldo_mipi_phy_config = {
        .chan_id = 3,
        .voltage_mv = 2500,
    };
    ESP_ERROR_CHECK(esp_ldo_acquire_channel(&ldo_mipi_phy_config, &ldo_mipi_phy));

    ESP_LOGI(TAG, "Initialize MIPI DSI bus");
    esp_lcd_dsi_bus_handle_t mipi_dsi_bus = NULL;
    esp_lcd_dsi_bus_config_t bus_config = DSI_PANEL_BUS_DSI_2CH_CONFIG();
    ESP_ERROR_CHECK(esp_lcd_new_dsi_bus(&bus_config, &mipi_dsi_bus));

    ESP_LOGI(TAG, "Install panel IO");
    esp_lcd_panel_io_handle_t mipi_dbi_io = NULL;
    esp_lcd_dbi_io_config_t dbi_config = DSI_PANEL_IO_DBI_CONFIG();
    ESP_ERROR_CHECK(esp_lcd_new_panel_io_dbi(mipi_dsi_bus, &dbi_config, &mipi_dbi_io));

    ESP_LOGI(TAG, "Install ESP_LCD panel driver");
    esp_lcd_panel_handle_t panel_handle = NULL;
    const esp_lcd_dpi_panel_config_t dpi_config = DSI_PANEL_DPI_CONFIG(LCD_COLOR_PIXEL_FORMAT_RGB888); // dunno maybe wrong format?
    dsi_vendor_config_t vendor_config = {
        .mipi_config = {
            .dsi_bus = mipi_dsi_bus,
            .dpi_config = &dpi_config,
        },
    };
    const esp_lcd_panel_dev_config_t panel_config = {
        .reset_gpio_num = -1,           // Set to -1 if not use
        .rgb_ele_order = LCD_RGB_ELEMENT_ORDER_RGB,     // Implemented by LCD command `36h`
        .bits_per_pixel = 24,    // Implemented by LCD command `3Ah` (16/18/24)
        .vendor_config = &vendor_config,
    };
    ESP_ERROR_CHECK(esp_lcd_new_panel_dsi(mipi_dbi_io, &panel_config, &panel_handle));
    ESP_ERROR_CHECK(esp_lcd_panel_reset(panel_handle));
    ESP_ERROR_CHECK(esp_lcd_panel_init(panel_handle));
    ESP_ERROR_CHECK(esp_lcd_panel_disp_on_off(panel_handle, true));
```

I then tried to simply set a pattern in the buffer to test the rendering to the display doing like:
```c
while(true) {
        esp_lcd_dpi_panel_set_pattern(panel_handle, MIPI_DSI_PATTERN_BAR_VERTICAL);
        ....
}
```

Could be that i simply missed something? 

The output i get is seemingly fine:
```log
I (4198) main_task: Started on CPU0
I (4218) main_task: Calling app_main()
I (4218) LCD: MIPI DSI PHY Powered on
I (4218) LCD: Initialize MIPI DSI bus
I (4218) LCD: Install panel IO
I (4228) LCD: Install ESP_LCD panel driver
I (4228) Waveshare DSI: version: 1.0.1
W (4238) i2c.master: Please check pull-up resistances whether be connected properly. Otherwise unexpected behavior would happen. For more detailed information, please read docs
I (4248) gpio: GPIO[7]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 0| Pulldown: 0| Intr:0 
I (4258) gpio: GPIO[8]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 0| Pulldown: 0| Intr:0 
I (4268) i2c_bus: i2c0 bus inited
I (4268) i2c_bus: I2C Bus V2 Config Succeed, Version: 1.2.0
I (4278) i2c_bus: i2c0 bus deinited
```

Some extra info i'm using ESP IDF 5.3.1 i flash the board via Serial. The LEDs on the Screen 4" Round display seem to go from a blinking state to a solid which at least on the RPI showed the display to be initialized.  

Could be also my revision of the silicon as i only have 0.1 version silicon (which i guess could have DSI issues) it seems the original screen (As in EK700x) is working without issues. If it is in fact a hardware issue then i'm also fine with it i didn't get informed by anyone that that would be the case.

Platin21, 2025-01-26:
Some update to this i was able to see a white line at the bottom of the screen when i invert the complete color of the screen.

Platin21, 2025-01-26:
[image]
[image]

Seems like some setting issue?

Platin21, 2025-01-26:
Another update after change of the hsync i was able to see the full pattern when setting it to 120 as in hsync_back_porch no color though.

Platin21, 2025-01-26:
Seems like a me issue on the side of the colors. Settings provided didn‘t work just to mention. 
I still have some artifact but that looks like another settings thing. 

[image]

Platin21, 2025-02-02:
It seems like the settings need to be different when using LVGL too? As otherwise one just gets blue screen.

WaveshareCloud, 2025-02-05:
Hi @Platin21 ,
Sorry for the late reply, I just finished my Chinese New Year holiday.
Ok, did you change the parameters in the esp_lcd_dsi.h file? Similar to the image below.
The specific screen after testing is no problem.
As for LVGL or other image display, then you need to know that the screen touch chip is GT9271, and its driver is basically the same as GT911, so you can use the driver of GT911. And to display content, yes, you have to use the esp_lcd_dsi library and then add lvgl panel to its panel by initializing lvgl
[image]

Platin21, 2025-02-05:
Hey, 

A happy new year then. 

yeah had to change them in order to get the DSI host rainbow. The normal params on my side resulted always in Blackscreen. I still got this Whiteline which i‘m not sure what its about. 

Ahh thanks for the hint about the touch driver. I didn‘t think that would be required. Will add a instance of the gt911 driver. My thinking was that LVGL allows one to call it without any input as it seemed to not require it. I guess i should get a picture in case i add the driver.

WaveshareCloud, 2025-02-05:
Hi @Platin21 ,
Thanks for your reply! 
If there are still issues with the software settings, I can combine a bin file for you to test. However, it might take a bit longer—you know how it is after a holiday, like a train station after rush hour, everything’s piled up and hectic!

WaveshareCloud, 2025-02-05:
We can first confirm that everything is working on the hardware before considering the software variant

Platin21, 2025-02-05:
Well on the hardware side i have the 4 pins connected GND, VCC, SDA, SCL. The DSI cable from the mipi dsi connector. The cable is connected on both ends i checked that. Other than that i don‘t know of anything. Tested it both with the Nano and ESP Kit same reaction with the driver.

WaveshareCloud, 2025-02-06:
You just need to connect the MIPI cable and power it

Platin21, 2025-02-06:
Well yeah that is already the case. 
Btw. i tried to use the GT911 driver and seems to make no diff. Or at least i can‘t seem to see more or less even when using it.

Y1hsiaochunnn, 2025-03-04:
Hi @Platin21 ,
Are you still here? So far I have tested the monitor driver on ESP32-P4-NANO, maybe you can watch the video below. Simply run an LVGL program combined with touch.

Nice flow. The following is a test program that runs on esp-idf master, but you should also be able to run it with idf v5.3 or higher

Demo code:[mipi_dsi_avoid_tearing.zip](https://github.com/user-attachments/files/19069854/mipi_dsi_avoid_tearing.zip)

[ESP32-P4-NANO-with-4inch_round_display.zip](https://github.com/user-attachments/files/19069833/ESP32-P4-NANO-with-4inch_round_display.zip)

WaveshareCloud, 2025-03-07:
There has been no response for a long time. If there is a problem, please open this issue or raise another one

closed 2025-03-07 by WaveshareCloud

Platin21, 2025-03-07:
Hey thanks for the example yeah seems to work fine with this. 
I have not look what i did wrong on my side that it failed. 
There is still some tearing though. I consider it being resolved though.

Y1hsiaochunnn, 2025-03-10:
Because you didn't change the screen resolution in the test_app main function, this causes the white lines to appear