waveshareteam/Waveshare-ESP32-components #79: 10.1" display flickering when many LVGL objects are on screen

Maldus512, 2025-08-29:
Hello,
I am developing a product using an ESP32P4 + 10.1" display [devkit](https://www.waveshare.com/wiki/ESP32-P4-Module-DEV-KIT-StartPage). Everything works quite smoothly except for some screen configurations with a lot of elements where the display flickers from time to time with seemingly no trigger. In the attached video there are three discernible flickers.

https://github.com/user-attachments/assets/0cade753-a169-4e4e-8179-a7d85d920da9

I've fiddled around and built a minimal example that reproduces the issue. It periodically prints the heap state because I believe the issue arises when some of the LVGL elements are allocated in the PSRAM instead of internal memory, but that's just an hypothesis. I am working with the latest branch of ESP-IDF v5.5.

[waveshare-flickering-report.zip](https://github.com/user-attachments/files/22041180/waveshare-flickering-report.zip)

Any suggestion on what could be the issue?

Y1hsiaochunnn, 2025-08-30:
I tried the code. Ok, so it's going to be blue as shown in the video. I think this is caused by the large size of the flushed lvgl block. Then it will scan the remaining psram and sram. So the blue screen. After I annotate the read memory capacity, the problem is solved.

closed 2025-08-30 by Y1hsiaochunnn

Y1hsiaochunnn, 2025-09-09:
[waveshare-flickering-report.zip](https://github.com/user-attachments/files/22230365/waveshare-flickering-report.zip)

Here's your code, and on top of that only comments what's read from memory.

Note that if everything is in PSRAM and you care about PSRAM rate, SRAM has rate limits as well. That's the point.

I think this part is more of a software issue, not hardware. Therefore, I recommend asking the professional technical team of ESpressif to get a better way to deal with it.