waveshareteam/Waveshare-ESP32-components #80: E (16272) lcd_panel.io.spi: panel_io_spi_tx_color(391): spi transmit (queue) color failed.(ESP32-S3-Touch-AMOLED-1.8)

yuexiajiayan, 2025-09-02:
#include "simple_wifi_sta.h"
 
 
 
 
//需要把这两个修改成你家WIFI，测试是否连接成功
#define DEFAULT_WIFI_SSID   "2hao"//"CMCC-irx2"
#define DEFAULT_WIFI_PASSWORD    "yibajiuling"  // "9mu7baav"
#define MAX_AP_NUM 10
#define TAG "wifi"
 
SemaphoreHandle_t* wifisem =NULL;
static char _ip_str[16];
char* wifi_sta_getip(){
    return _ip_str; 
}
char* getcssid(){
    return "1hao";
}
/** 事件回调函数
 * @param arg   用户传递的参数
 * @param event_base    事件类别
 * @param event_id      事件ID
 * @param event_data    事件携带的数据
 * @return 无
*/
static void event_handler(void* arg, esp_event_base_t event_base,int32_t event_id, void* event_data)
{   
    if(event_base == WIFI_EVENT)
    {
        switch (event_id)
        {
        case WIFI_EVENT_STA_START:      //WIFI以STA模式启动后触发此事件
            esp_wifi_connect();         //启动WIFI连接
            break;
        case WIFI_EVENT_STA_CONNECTED:  //WIFI连上路由器后，触发此事件
            ESP_LOGI(TAG, "connected to AP");
            
            break;
        case WIFI_EVENT_STA_DISCONNECTED:   //WIFI从路由器断开连接后触发此事件
            esp_wifi_connect();             //继续重连  需要延迟
            ESP_LOGI(TAG,"connect to the AP fail,retry now");
            break;
        default:
            break;
        }
    }
    if(event_base == IP_EVENT)                  //IP相关事件
    {
        switch(event_id)
        {
            case IP_EVENT_STA_GOT_IP:           //只有获取到路由器分配的IP，才认为是连上了路由器
                ESP_LOGI(TAG,"get ip address");
                if(wifisem!=NULL){
                   
                    // 获取 IP 地址信息
                    esp_netif_ip_info_t ip_info;
                    esp_netif_t *netif = esp_netif_get_handle_from_ifkey("WIFI_STA_DEF");
                     
                    esp_netif_get_ip_info(netif, &ip_info);
 
                    // 转换为字符串格式 
                    inet_ntoa_r(ip_info.ip.addr, _ip_str, sizeof(_ip_str));
                    //char mask_str[16];
                    // inet_ntoa_r(ip_info.netmask.addr, mask_str, sizeof(mask_str));
                    // char gw_str[16];
                    //inet_ntoa_r(ip_info.gw.addr, gw_str, sizeof(gw_str));
 
                   
                    // 或者直接使用 LWIP 的 ip4addr_ntoa 函数
                    //ESP_LOGI(TAG, "STA got IP address: %s", ip4addr_ntoa(&ip_info.ip));
                    // INET_ADDRSTRLEN 定义了存储 IPv4 地址所需的字符数 
                    xSemaphoreGive(*wifisem); 
                    
                }
                break;
        }
    }
}
 
 
//WIFI STA初始化
esp_err_t wifi_sta_init(SemaphoreHandle_t* sem)
{   
    wifisem = sem;
    ESP_ERROR_CHECK(esp_netif_init());  //用于初始化tcpip协议栈
    ESP_ERROR_CHECK(esp_event_loop_create_default());       //创建一个默认系统事件调度循环，之后可以注册回调函数来处理系统的一些事件
    esp_netif_create_default_wifi_sta();    //使用默认配置创建STA对象
 
    //初始化WIFI
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    
    //注册事件
    ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT,ESP_EVENT_ANY_ID,&event_handler,NULL));
    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT,IP_EVENT_STA_GOT_IP,&event_handler,NULL));
 
    //WIFI配置
    // wifi_config_t wifi_config = 
    // { 
    //     .sta = 
    //     { 
    //         .ssid = DEFAULT_WIFI_SSID,              //WIFI的SSID
    //         .password = DEFAULT_WIFI_PASSWORD,      //WIFI密码
    //         .threshold.authmode = WIFI_AUTH_WPA2_PSK,   //加密方式
            
    //         .pmf_cfg = 
    //         {
    //             .capable = true,
    //             .required = false
    //         },
    //     },
    // };
    nvs_handle_t nvs_handle;
    wifi_config_t wifi_config;
 
    esp_err_t  ret = nvs_open("storage", NVS_READONLY, &nvs_handle);
    if (ret == ESP_OK) {
        size_t ssid_length = sizeof(wifi_config.sta.ssid);
        size_t password_length = sizeof(wifi_config.sta.password);
        
        ret = nvs_get_str(nvs_handle, "wifi_ssid", (char*)wifi_config.sta.ssid, &ssid_length);
        if (ret != ESP_OK) {
            // 如果没有找到SSID，使用默认值
            strncpy((char*)wifi_config.sta.ssid, DEFAULT_WIFI_SSID, sizeof(wifi_config.sta.ssid));
        }
 
        ret = nvs_get_str(nvs_handle, "wifi_password", (char*)wifi_config.sta.password, &password_length);
        if (ret != ESP_OK) {
            // 如果没有找到密码，使用默认值
            strncpy((char*)wifi_config.sta.password, DEFAULT_WIFI_PASSWORD, sizeof(wifi_config.sta.password));
        } 
        nvs_close(nvs_handle);
        ESP_LOGE(TAG,"loading old wifi");
    } else {
        // 如果无法打开NVS，使用默认值
        strncpy((char*)wifi_config.sta.ssid, DEFAULT_WIFI_SSID, sizeof(wifi_config.sta.ssid));
        strncpy((char*)wifi_config.sta.password, DEFAULT_WIFI_PASSWORD, sizeof(wifi_config.sta.password));
          ESP_LOGE(TAG,"loading default wifi");
    }
 
    // 设置加密方式等其他配置
    wifi_config.sta.threshold.authmode = WIFI_AUTH_WPA2_PSK;
    wifi_config.sta.pmf_cfg.capable = true;
    wifi_config.sta.pmf_cfg.required = false;
    
    //启动WIFI
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA) );         //设置工作模式为STA
    ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_STA, &wifi_config) );   //设置wifi配置
    ESP_ERROR_CHECK(esp_wifi_start() );                         //启动WIFI
    
    ESP_LOGI(TAG, "wifi_init_sta finished.");
    return ESP_OK;
}
 
wifi_ap_record_t* wifi_scan(uint16_t *out_ap_num) {
    // 配置扫描参数
    wifi_scan_config_t scanConf = {
        .ssid = NULL,
        .bssid = NULL,
        .channel = 0,   // 0 表示扫描所有通道
        .show_hidden = true,
        .scan_type = WIFI_SCAN_TYPE_ACTIVE,
        .scan_time = {.active = {.min = 0, .max = 0}} // 使用默认值
    };
 
    // 开始扫描
    ESP_ERROR_CHECK(esp_wifi_scan_start(&scanConf, true));
 
   wifi_ap_record_t *ap_info = calloc(MAX_AP_NUM, sizeof(wifi_ap_record_t));
    if (ap_info == NULL) {
        ESP_LOGE(TAG, "Failed to allocate memory for AP records");
        return NULL;
    }

    uint16_t ap_num = MAX_AP_NUM;
    esp_err_t err = esp_wifi_scan_get_ap_records(&ap_num, ap_info);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Failed to get AP records: %s", esp_err_to_name(err));
        free(ap_info);
        return NULL;
    }

    // 获取总探测数（调试用）
    uint16_t total_count;
    ESP_ERROR_CHECK(esp_wifi_scan_get_ap_num(&total_count));
    ESP_LOGI(TAG, "Total APs scanned = %u", total_count);

    //  正确设置输出参数
    if (out_ap_num != NULL) {
        *out_ap_num = ap_num;  //  解引用赋值
    }

    // 可选：打印结果
    for (int i = 0; i < ap_num; i++) {
        ESP_LOGI(TAG, "SSID: %s, RSSI: %d dBm, Channel: %d", 
                 ap_info[i].ssid, ap_info[i].rssi, ap_info[i].primary);
    }

    //  返回堆内存指针（调用者需 free）
    return ap_info;
}

============

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "nvs_flash.h"
#include "nvs.h"
#include "esp_log.h"
#include "esp_err.h"
#include "esp_check.h"
#include "esp_memory_utils.h"
#include "lvgl.h"
#include "bsp/esp-bsp.h"
#include "bsp/display.h"
#include "lv_demos.h"

SemaphoreHandle_t wifi_sem; 

void app_main(void)
{
    //init wifi
    wifi_sem =xSemaphoreCreateBinary();
      //NVS初始化（WIFI底层驱动有用到NVS，所以这里要初始化）
    nvs_flash_init();
    //wifi STA工作模式初始化
    wifi_sta_init(&wifi_sem);
  xSemaphoreTake(wifi_sem,portMAX_DELAY );

 
    bsp_display_start();

    bsp_display_lock(0);
    bsp_display_brightness_set(60); 
    bsp_sdcard_mount();

    // lv_demo_music();
    // lv_demo_benchmark();
     lv_demo_widgets();

    bsp_display_unlock();
}

当添加wifi的初始化就提示该错误，是需要调整什么参数吗在ESP32-S3-Touch-AMOLED-1.8上

yuexiajiayan, 2025-09-03:
@Y1hsiaochunnn @WaveshareCloud

Y1hsiaochunnn, 2025-09-04:
nvs_flash_init放最前面。WiFi的函数对吗？看的idf的示例？

Y1hsiaochunnn, 2025-09-04:
问题肯定不会出现在驱动层，肯定是WIFI的函数哪里进入了死循环。我这里也有连接WIFI然后运行lvgl可以触摸的示例，也没有上述的问题。

[Pixel-Clock.zip](https://github.com/user-attachments/files/22135466/Pixel-Clock.zip)